prusa3d/Prusa-Link-Web #26: Estimated End show only current time

ingsme, 2020-09-30:
I can connect to my Prusa Mini with a browser and while it currently only shows temperature and some other statistics I expect this to improve over time.
However, on the Dashboard the estimated time show only current time, no estimation.

Chleba, 2020-10-01:
Hi @ingsme,
Can you please wrote me Your version of firmware ?
Thank you.

DRracer, 2020-10-02:
@ingsme can you post the G-code you were printing? Was it sliced in PrusaSlicer or something else?

ingsme, 2020-10-02:
Yes it was sliced in PrucaSlicer.

```
; generated by PrusaSlicer 2.2.0+linux-x64 on 2020-09-30 at 14:07:07 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.70mm

M107
M104 S200 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M109 S200 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
G1 Z0.350 F7800.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X78.866 Y55.656 F7800.000
G1 E2.00000 F2400.00000
G1 F1800.000
G1 X80.207 Y54.587 E2.15597
…
G92 E0
M107
; Filament-specific end gcode 
;END gcode for filament
M104 S0 ; turn off temperature
G28 X0  ; home X axis
M84     ; disable motors
; filament used [mm] = 9444.5
; filament used [cm3] = 22.7; printer_settings_id = - default FFF -
; printer_technology = FFF
; printer_variant = 
; printer_vendor = 
; raft_layers = 0
; remaining_times = 0
; resolution = 0
; retract_before_travel = 2
; retract_before_wipe = 0%
; retract_layer_change = 0
; retract_length = 2
; retract_length_toolchange = 10
; retract_lift = 0
; retract_lift_above = 0
; retract_lift_below = 0
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 40
; seam_position = aligned
; serial_port = 
; serial_speed = 250000
; silent_mode = 1
; single_extruder_multi_material = 0
; single_extruder_multi_material_priming = 1
; skirt_distance = 6
; skirt_height = 1
; skirts = 1
; slice_closing_radius = 0.049
; slowdown_below_layer_time = 5
; small_perimeter_speed = 15
[... 261 lines cut ...]
; top_fill_pattern = rectilinear
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 15
; top_solid_layers = 3
; top_solid_min_thickness = 0
; travel_speed = 130
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; variable_layer_height = 1
; wipe = 0
; wipe_into_infill = 0
; wipe_into_objects = 0
; wipe_tower = 0
; wipe_tower_bridging = 10
; wipe_tower_no_sparse_layers = 0
; wipe_tower_rotation_angle = 0
; wipe_tower_width = 60
; wipe_tower_x = 180
; wipe_tower_y = 140
; wiping_volumes_extruders = 70,70
; wiping_volumes_matrix = 0
; xy_size_compensation = 0
; z_offset = 0
```

ingsme, 2020-10-04:
This may be a false alarm. I notice that the status on the Prusa display do not give a "Remaining Time" on this print either. I wonder why though. The piece have been sliced in PrusaSlicer and in the slicer it estimates a printing time.

However. That is probably a problem for another project, not this one.

DRracer, 2020-10-05:
@ingsme thank you ... the most interesting part is missing though - do you see any M73 commands in your G-code? How do they look like?

ingsme, 2020-10-05:
No M73 commands, no. This seem to be an issue with the linux version of the PrucaSlicer. I just tested on Mac and Linux. The gcode produced on the Mac has both M73 commands and thumbnail while the gcode produced on linux do not.

ingsme, 2020-10-05:
I can confirm that the problem was the Appimage version of PrusaSlicer I was using. On a PrusaSlicer I compiled on Linux myself there was no problem with the gcode and the Estimated End was working as it should.

closed 2020-10-05 by ingsme