prusa3d/Prusa-Link-Web #34: FR: Expose metrics using Prometheus text format

arnauldvm, 2020-11-06:
This is a Feature Request.

As far as I found, metrics exposed by the printer (cf. /api/telemetry endpoint) use a JSON format.

This does not make possible to integrate easily with existing metrics stacks, like [Prometheus](https://prometheus.io/).

Prometheus is one of the most popular metrics stack (yet not the only one). More interestingly, its [text based exposition format](https://prometheus.io/docs/instrumenting/exposition_formats/) is becoming a de facto standard, and is quite easy to implement.

Integrating a metrics stack with (farm of) printers could be valuable both for hobbyists and professionals, since it would provide, almost out-of-the-box, with monitoring dashboards and automatic alerting.

Note that this is not a strong feature requests, since there exists workarounds, like using the community [https://github.com/prometheus-community/json_exporter](https://github.com/prometheus-community/json_exporter) to convert JSON to Prometheus format. It's just one more hop in the stack.

Though, promoting such integrations could bring interesting developments.

ondratu, 2020-11-06:
Hi, thanks for your issue.

Could you tell me please, what kind of information do you want to see in Prometheus stats? I know this stats only for servers, and mainly they are counters there. But in /api/telemetry that are actual printing values, compatible to Octoprint API. So if you need another format, that must be implement as second endpoint.

So could you tell me, how this data can look?

RichiH, 2020-11-27:
Prometheus maintainer here, just got my Mini yesterday. I still hope it's a Mini+ but don't know how to determine if it's a SuperPINDA, yet.

https://plugins.octoprint.org/plugins/prometheus_exporter/ might also help here, but native metrics would be better.

Now that we have released OpenMetrics, implementing this would be better and it almost looks the same https://github.com/OpenObservability/OpenMetrics/blob/master/specification/OpenMetrics.md

A line like 

`{"temp_nozzle":240,"temp_bed":90,"material":"PETG","pos_z_mm":13.82,"printing_speed":100,"flow_factor":95,"progress":77,"print_dur":"      3h 55m","time_est":"3960","project_name":"CleatHaakKlem2_0.2mm_PETG_MINI_4h59m.gcode"}`

would turn into something like this:

````
# TYPE prusa_info info
prusa_info{printer="mini",revision="plus"} 1
# TYPE prusa_print_info info
prusa_print_info{material="PETG",print_name="CleatHaakKlem2_0.2mm_PETG_MINI_4h59m.gcode"} 1
# TYPE prusa_nozzle_temperature_celsius gauge
# UNIT prusa_nozzle_temperature_celsius celsius
prusa_temperature_celsius{sensor="bed"} 90
prusa_temperature_celsius{sensor="nozzle"} 240
# TYPE prusa_target_temperature_celsius gauge
# UNIT prusa_target_temperature_celsius celsius
prusa_target_temperature_celsius{sensor="bed"} 90
prusa_target_temperature_celsius{sensor="nozzle"} 240
# TYPE prusa_position_meter gauge
# UNIT prusa_position_meter meter
prusa_position_meter{axis="x"} 0.123
prusa_position_meter{axis="y"} 0.123
prusa_position_meter{axis="z"} 0.01382
# TYPE prusa_print_target_time_seconds gauge
# UNIT prusa_print_target_time_seconds seconds
# HELP prusa_print_target_time_seconds Total time the print will take
prusa_print_target_time_seconds 10000
# TYPE prusa_print_seconds counter
# UNIT prusa_print_seconds seconds
# HELP prusa_print_seconds seconds Time print has taken up to now
prusa_print_seconds 7700
EOF
````

`printing_speed` and `flow_factor` not shown as I don't know the units, but those can be done similarly.

RichiH, 2020-11-28:
Another useful class of metrics would be health & alert states, e.g. if the filament sensor is detected runout or if a temperature alert is firing.

RichiH, 2020-11-28:
To compare, this what comes out of the Prometheus plugin:

````
# HELP octoprint_print_time_elapsed Print time elapsed
# TYPE octoprint_print_time_elapsed gauge
octoprint_print_time_elapsed{path="Spool_holder_-_Spool_holder_0.2mm_PETG_MINI_2h22m.gcode"} 3272.0
# HELP octoprint_print_time_est Print time estimate
# TYPE octoprint_print_time_est gauge
# HELP octoprint_print_time_left_estimate Print time left estimate
# TYPE octoprint_print_time_left_estimate gauge
octoprint_print_time_left_estimate{path="Spool_holder_-_Spool_holder_0.2mm_PETG_MINI_2h22m.gcode"} 11867.0
# HELP octoprint_temperatures_actual Reported temperatures
# TYPE octoprint_temperatures_actual gauge
octoprint_temperatures_actual{identifier="B"} 90.03
octoprint_temperatures_actual{identifier="T0"} 240.0
# HELP octoprint_slice_progress Slice progress
# TYPE octoprint_slice_progress gauge
# HELP octoprint_raspberry_core_temperature Core temperature of Raspberry Pi
# TYPE octoprint_raspberry_core_temperature gauge
octoprint_raspberry_core_temperature 0.0
# HELP octoprint_printing_time_total Printing time total
# TYPE octoprint_printing_time_total counter
octoprint_printing_time_total 0.0
# HELP octoprint_printing_time_created Printing time total
# TYPE octoprint_printing_time_created gauge
octoprint_printing_time_created 1.606563107690349e+09
# HELP octoprint_infos_info Octoprint host informations
# TYPE octoprint_infos_info gauge
octoprint_infos_info{app_start="1606563123",host="octopi",octoprint_version="1.4.2",platform="Linux"} 1.0
# HELP octoprint_client_num The number of connected clients
# TYPE octoprint_client_num gauge
octoprint_client_num 1.0
# HELP octoprint_extrusion_total Filament extruded total
# TYPE octoprint_extrusion_total counter
octoprint_extrusion_total 3553.791089999683
# HELP octoprint_extrusion_created Filament extruded total
# TYPE octoprint_extrusion_created gauge
octoprint_extrusion_created 1.606563107690537e+09
# HELP octoprint_temperatures_target Targeted temperatures
# TYPE octoprint_temperatures_target gauge
octoprint_temperatures_target{identifier="B"} 90.0
octoprint_temperatures_target{identifier="T0"} 240.0
# HELP octoprint_started_prints_total Started print jobs
# TYPE octoprint_started_prints_total counter
octoprint_started_prints_total 1.0
# HELP octoprint_started_prints_created Started print jobs
# TYPE octoprint_started_prints_created gauge
octoprint_started_prints_created 1.606563107688677e+09
# HELP octoprint_extrusion_print Filament extruded this print
# TYPE octoprint_extrusion_print gauge
octoprint_extrusion_print 3553.791089999683
# HELP octoprint_x_travel_total X axis travel total
# TYPE octoprint_x_travel_total counter
octoprint_x_travel_total 70539.27500000423
# HELP octoprint_x_travel_created X axis travel total
# TYPE octoprint_x_travel_created gauge
octoprint_x_travel_created 1.606563107690857e+09
# HELP octoprint_printer_state_info Printer connection info
# TYPE octoprint_printer_state_info gauge
octoprint_printer_state_info{state_id="PRINTING",state_string="Printing"} 1.0
# HELP octoprint_x_travel_print X axis travel in this print
# TYPE octoprint_x_travel_print gauge
octoprint_x_travel_print 70539.27500000423
# HELP octoprint_done_prints_total Done print jobs
# TYPE octoprint_done_prints_total counter
octoprint_done_prints_total 0.0
# HELP octoprint_done_prints_created Done print jobs
# TYPE octoprint_done_prints_created gauge
octoprint_done_prints_created 1.606563107689015e+09
# HELP octoprint_y_travel_total Y axis travel total
# TYPE octoprint_y_travel_total counter
octoprint_y_travel_total 70016.16100000062
# HELP octoprint_y_travel_created Y axis travel total
# TYPE octoprint_y_travel_created gauge
octoprint_y_travel_created 1.606563107691176e+09
# HELP octoprint_captured_timelapses_total Timelapse captured
# TYPE octoprint_captured_timelapses_total counter
octoprint_captured_timelapses_total 0.0
# HELP octoprint_captured_timelapses_created Timelapse captured
# TYPE octoprint_captured_timelapses_created gauge
octoprint_captured_timelapses_created 1.60656310768934e+09
# HELP octoprint_y_travel_print Y axis travel in this print
# TYPE octoprint_y_travel_print gauge
octoprint_y_travel_print 70016.16100000062
# HELP octoprint_failed_prints_total Failed print jobs
# TYPE octoprint_failed_prints_total counter
octoprint_failed_prints_total 0.0
# HELP octoprint_failed_prints_created Failed print jobs
# TYPE octoprint_failed_prints_created gauge
octoprint_failed_prints_created 1.606563107688855e+09
# HELP octoprint_z_travel_total Z axis travel total
# TYPE octoprint_z_travel_total counter
octoprint_z_travel_total 105.60000000000096
# HELP octoprint_z_travel_created Z axis travel total
# TYPE octoprint_z_travel_created gauge
octoprint_z_travel_created 1.606563107692632e+09
# HELP octoprint_z_travel_print Z axis travel in this print
# TYPE octoprint_z_travel_print gauge
octoprint_z_travel_print 105.60000000000096
# HELP octoprint_print_fan_speed Fan speed
# TYPE octoprint_print_fan_speed gauge
octoprint_print_fan_speed 124.95
# HELP octoprint_cancelled_prints_total Cancelled print jobs
# TYPE octoprint_cancelled_prints_total counter
octoprint_cancelled_prints_total 0.0
# HELP octoprint_cancelled_prints_created Cancelled print jobs
# TYPE octoprint_cancelled_prints_created gauge
octoprint_cancelled_prints_created 1.606563107689174e+09
# HELP octoprint_print_progress Print progress
# TYPE octoprint_print_progress gauge
octoprint_print_progress{path="Spool_holder_-_Spool_holder_0.2mm_PETG_MINI_2h22m.gcode"} 20.0
````

mmanjos, 2024-02-15:
Just came across this request - it would be great to see this supported upstream.

@RichiH 's solution to use json_exporter is probably a better answer long-term but I wanted to give writing my first custom exporter a try and just I got my Prusa XL delivered last weekend.

I put this project together: https://github.com/mmanjos/prometheus-prusalink-exporter

Hopefully it's helpful to others out there

RichiH, 2024-02-15:
I had forgotten this issue existed, it had been unactioned so long..

Thanks for the link! Also take a look at https://github.com/pstrobl96/prusa_exporter by @pstrobl96 - I've been using it successfully at home for some time now and we use it at conferences at which Grafana has a booth. We usually bring a Mk4 these days to have something tangible for people to see Prometheus & Grafana in action with.