prusa3d/Prusa-Link-Web #43: Wrong (different) font  on language dropdown

Gordontgopher, 2021-01-18:
[image]
It appears that the font on the drop down is different to the rest of the screen.

This is from a Mini running 4.3.0 RC2 firmware.

Gordontgopher, 2021-07-28:
Where should this issue be logged now? I opened it against Prusa Connect, which has now been renamed to Prusa-Link-Web. Should I just not bother moving it to the correct repository as nobody has acknowledged it in 6 months?

Gordontgopher, 2021-09-02:
@JohnnyDeer Is this in the wrong place or just being ignored due to higher priority on Mk3 PrusaConnect?

ondratu, 2022-05-16:
This bug was fixed yet, and all printers will release bug fix (some was) in next public release.

closed 2022-05-16 by ondratu