prusa3d/Prusa-Link-Web #159: Option to cancel transfer

tomekmgr, 2022-05-03:
There should be an option to cancel transfer or at least to reboot the Rpi from web interface.

I have power on the Printer, and from Prusa Slicer send a gcode to upload and print.
After a while Prusa Slicer show an errror, that transfer failed.
On the UI in Prusa Link file transfer stops at 99.9% and nothing is happening.

Second attempt to send a gcode end with an error that transfer is already in progress.
There is no way to stop this remotely, cancel print button does not work.
Cannot login via SSH to Rpi, looks like SSH is disabled on prusa link image.

The only  solution is AC cycle the printer, which is not really convenient way 

Setup is MK3S+ with Rpi Zero.
Versions:
```
api 2.0.0
firmware 3.11.0-4955
server 0.6.1
text OctoPrint 1.1.0
sdk 0.6.0
frontend 3.5.0

```

ondratu, 2022-05-16:
Thanks for your bug report. Yes there was a bug in Link, but it was fixed, and will be released in next public release.

But your request to cancel transfer in PrusaLink or in PrusaConnect is ok, and we will do it.

tomekmgr, 2022-05-16:
Good to hear that, thanks.

LostinAge, 2023-07-19:
I see that the feature was addressed I have set my system up with Prusalink and the issue seems to be there. The file transfer is stuck at 10% and I cannot get to cancel the transfer. I transferred 3 other files before it stuck. What is the version I need to get and what is the command?