prusa3d/Prusa-Link-Web #161: [Feature Request] Settings page to change hostname

Danil0v3s, 2022-05-09:
So I've been talking with @TojikCZ on Discord and the idea of creating an electron app that would offload everything from the printers board and give the ability to have more contributions etc came up and so I built this https://github.com/Danil0v3s/Prusa-Link-Standalone/

For that app to work properly we would need the ability to change the hostname externally.. I tried to trace where does the hostname comes from and it looks like it comes from a hardcoded call to `localhost`. Is that correct?

What would be the best approach in your opinion to tackle this issue if this is even something that could be included in the roadmap. 

IMHO, I think it's a cool feature

TojikCZ, 2022-05-10:
I asked abut the Idea of having an external app. Most of the Icons are shared with the internal GUI, so the JS on top is not really the memory hog I thought it was. That mostly ruins my argument about not wanting to add a terminal because it would cost more bytes. Still, I don't think the terminal will get added anytime soon if ever. Please don't put more work on the app :/ for the time being it looks like the web interface is the solution we want to use. Sorry

ondratu, 2022-05-10:
At this moment we are hard working on new web interface, which I means:
* new Open API endpoints and new data - we are working on more intuitive and usable endpoints and data models - for example only one telemetry endpoint not more
* new Web UI, which will use new API, and do some new - better functionality

About your request to changing hostname - yes, at this moment, we have some endpoints to change some attributes, but that is only beginning. But on PrusaLink for MK3, many of system attributes are so system settings. And we must make some bridge to low system API. SL1 has this bridge, MINI and MK3 not yet.