prusa3d/Prusa-Link-Web #198: Setup error: "Can't connect to Prusa Connect."

mathieucarbou, 2022-06-26:
This is a copy of https://forum.prusa3d.com/forum/bugs-errors/error-cant-connect-to-prusa-connect/#post-611598 with more info.

[image]

I I can SSH into the device and I can reach the host though...

```shell
pi@prusa:~ $ nc -v -z connect.prusa3d.com 443
Connection to connect.prusa3d.com (172.104.231.244) 443 port [tcp/https] succeeded!
```

* I have the latest firmware: prusa3d_fw_3_11_0_MK3S
* Printer: Original Prusa i3 MK3S+
* Raspberry Pi Zero W running PrusaLink downloaded from the link at the official page here: https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-for-mk3-s-_221744
* Image version is 0.6.1

Please see attached logs from: `prusa-link -f -d start`: [debug.log](https://github.com/prusa3d/Prusa-Link-Web/files/8987708/debug.log)

Prusa-Support, 2022-07-28:
Thanks for reaching us.

Apparently, there is a problem with connecting Prusa Link to Prusa Connect.
Some reasons for the problem could be, for example:
- problem with DNS;
- some PROXY on the way;
- some other (IP routing) problem;

and more...

We are working on explanatory error messages, which will be implemented in the next update (coming soon).
You can try reviewing one of the listed examples of issues, or please wait for the next version.
Thanks for your patience.

Michele Moramarco
Prusa Research

ondratu, 2022-10-19:
Is that error still current? Can we close this issue? @mathieucarbou

mathieucarbou, 2022-10-19:
Yes we can close.

Prusa-Support, 2022-10-19:
Thank you for your cooperation and I'll close this issue now.

closed 2022-10-19 by Prusa-Support