prusa3d/Prusa-Link-Web #212: PrusaLink API Error

4ever-audi, 2022-07-29:
Hi

I have one on my Prusa Mini
Wifi module installed and installed the new firmware.
Has everything worked so far, I can also directly from
Slicer upload the file to the printer.
Everything is ok until leveling, but then when it starts printing the website writes "API Error".
Can you help me?

Thank you LG

Prusa-Support, 2022-08-31:
Thank you for reaching us.

Not sure if this issue concerns the MINI firmware or Prusa Link.

Does the error only show up on the Prusa Link webpage, without preventing the printer from printing?

Please perform a [factory reset](https://help.prusa3d.com/article/factory-reset-mini_177516) and repeat the wi-fi configuration as per the [instructions](https://help.prusa3d.com/guide/how-to-setup-wi-fi-mini-mini_316781).

Would the error show up with all sent G-codes, or with a specific one?
Would you mind providing a problematic G-code for us to try to reproduce the issue?

Michele Moramarco
Prusa Research

Prusa-Support, 2022-10-11:
Possibly in the near future, with the updates of MINI firmware and Prusa Link, the chances of encountering such errors will be reduced, or these will be more easily addressed.
This issue will be closed now, as unable to be reproduced, hence unconfirmed.

Michele Moramarco
Prusa Research

closed 2022-10-11 by Prusa-Support