prusa3d/Prusa-Link-Web #217: Add the ability to upload multiple gcodes

Underwater-Hockey, 2022-08-04:
Currently, I can upload a single gcode and over the current network implementation for wifi, this takes a while. 
I'd like the ability to select multiple files for upload so that I don't have to perform this step manually for each file.

Prusa-Support, 2022-09-01:
Thank you for your suggestion.

It is possible to create a queue of uploads in Prusa Slicer, if you freshly slice and upload several print files quickly, one after another.
Uploading multiple files is not possible in Prusa Link at the moment and I'm not sure will be possible in the future.
I'd leave this issue open and let the developers think about it.

Nonetheless, in the future, Prusa Connect should have features to manage multiple print files and multiple printers. Maybe an option like that will be available when Prusa Connect will be fully operative.

Michele Moramarco
Prusa Research

ondratu, 2022-10-19:
PrusaLink probably never allowed upload more files at once, but Connect can do that. At this moment, we are working on transfer queue on Connect side.

closed 2022-10-19 by ondratu