prusa3d/Prusa-Link-Web #222: [ENHANCEMENT] Make API key entry a password field and/or set an authentication cookie

RichiH, 2022-08-11:
Prusa Mini, Filament sensor & ESP01S installed, v4.4.0-beta2

Connecting to Prusa Link Beta requires an API key through a clear text field. Browser do not autodetect this as a password field, so they do not offer to save the API key. Neither does Prusa Link Beta set a cookie with the API key.
This means that I have to look up and enter the API key every single time.

For more convenience, please
1. make the API key entry a password field, and/or
2. set an authentication cookie.

Prusa-Support, 2022-08-13:
Thank you for your suggestion.

Our developers will look into this.
Also, this behavior may change when the MINI will be added to Prusa Connect.

Michele Moramarco
Prusa Research

mix579, 2022-08-26:
+1. Very irritating

mix579, 2022-08-27:
Just to clarify, as the issue [#2428](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2428#issuecomment-1228902285) was closed as related:

I see this as two separate issues.

First, the suggestion to turn the browser field from clear text into a password field. Now password managers can autofill it. However, without addressing suggestion 2 I would still get prompted every time for the API key.

Suggestion 2 (implementing a cookie) would address being prompted each time but would not allow a password manager to autofill the field. So the two ideas are closely related but not identical.

When I posted #2428, I was concerned about being prompted every time and didn't even think about the password field idea. Obviously I am supportive of a solution to BOTH of these issues.

ondratu, 2022-10-19:
I was close that issue because the API type login in PrusaLink Web was temporary. In next release, only HTTP Digest will be allowed.

So i closed that issue, it is solved at this moment for all our printers. Please wait for next firmware for MINI.

RichiH, 2022-10-20:
Thanks @ondratu !

Is there a release date for the alpha / beta / full release?

Prusa-Support, 2022-10-29:
Hard to tell @RichiH, but we already got the first release candidate (= almost stable) [Firmware-Buddy v4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1).
The next release shouldn't be far, it's a matter of time. 🙂 

Michele Moramarco
Prusa Research

mix579, 2022-10-29:
Maybe I'm misunderstanding something here but I'm running v4.40-RC1, and I still get prompted for the API key every time. Nor is it recognized as a password field by 1Password? So I don't understand why it's closed.

Prusa-Support, 2022-11-04:
> I was close that issue because the API type login in PrusaLink Web was temporary. In next release, only HTTP Digest will be allowed.
> 
> So i closed that issue, it is solved at this moment for all our printers. Please wait for next firmware for MINI.

[Firmware-Buddy v4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1) release is prior to this https://github.com/prusa3d/Prusa-Link-Web/issues/222#issuecomment-1283584841. Please wait for the next release.

Michele Moramarco
Prusa Research

GithubUser99999999, 2022-11-27:
Now after v4.4.0 final has been released, this still hasn't changed and I have to enter a cryptic string of characters EVERY SINGLE TIME I want to use PrusaLink. Why did nobody ever try to actually use this so far?

mix579, 2022-11-27:
I can confirm that. Wasn't that supposed to be fixed in the final release? At least now—for some reason—1Password seems to recognize it so I can use it to enter the password with one click.

KlausKraemer, 2022-11-28:
As a private user I have all but no use for an API key. Memorizing these keys isn't possible if you have no eidetic brain. So I'm f***d when I want to access my PRUSA from a tablet or a Computer not in the vicinity of my printer. It's really very impractical.

Please make it simple and let users chose themselves:

- disable API key (=never ask for one)
- Set an API key via PrusaLink surface (so you could chose a password you can reasonably memorize)
- the existing option (for those who like or need it)

Entering the API key via a password input would be also convenient to use a password manager.

Thanks!

RichiH, 2022-11-29:
@Prusa-Support @ondratu unfortunately this is not actually resolved in the current firmware. Could you please re-open this issue?

ondratu, 2022-11-29:
Will be fixed in new 4.5.x release

MartinZubek, 2022-12-11:
> Will be fixed in new 4.5.x release

So you basically mean we'll have to wait another year for it? 😮

If I am looking correctly it takes about a year to release a minor version update (4.3.4 was released in December 2021 and 4.4 in November 2022). Let's hope there is something planned sooner than that.

remys83, 2022-12-12:
I hope 4.5.X is released ASAP to resolve this or the fix is included in a sooner 4.4.X version. I'm tempted to roll back the firmware because this is very inconvenient to enter the API each time. The only way I've gotten around this is to not close the browser tab and the link is maintained (for awhile).

guitar24t, 2023-01-28:
I'm disappointed to see that this wasn't fixed in the 4.4.1 firmware. Is there an eta on 4.5.x firmware release where this issue will be resolved? We use many minis with PrusaLink local in a print farm environment for our shop and having to type the API key every time is awful. We want to be able to disable it or at the very least have our browsers remember it. I would happily compile a custom firmware for this machine without the feature enabled, however, I don't want to void our warranties.

KlausKraemer, 2023-01-28:
Promises, promises....

Get rid of that nagging, mandatory API-key forced also on users that never need it!

ondratu, 2023-01-31:
It is easy. Our new code, which is in repo didn't accept Api-Key as login credential any more. And we focus to new firmware version. Fixing any old code, which is deleted yet, creates new delay to releasing new firmware. Fixes of old firmware (like 4.4.1) is about printing.

Please, stay tuned. We are working hard on new features for 4.5.0, which increase printer connectivity services.

mix579, 2023-01-31:
Happy to hear this.

Kinda related: are there ANY plans to integrate the Mini's PrusaLink implementation into PrusaConnect. The Mini was touted originally as the "ideal farm" printer, and now the Minis are the only printers in my arsenal I can't control as part of the farm with PrusaConnect.

Prusa-Support, 2023-02-09:
Yes, MINI will be made compatible with Prusa Connect.
Unfortunately, we don't have a time estimation for that either.

Michele Moramarco
Prusa Research

jagr200, 2023-03-18:
Imagine thinking an API key is necessary for a 3D printer.... This is awful. What a terrible service yet again.

MartinZubek, 2023-03-23:
I'm still hoping that some kind of fix for this is just around the corner.. 🙁

Imagine you have to enter the key several times a day for your home printer. Even when it is just copy-paste, it's still tedious.
At least if there was some way how to save the key to that field, but since the field itself is not a password field, even the browser can't help with this here.

Come on guys, I don't believe that this is the way you want us have it.
And please don't let us wait several more months until some next major release, please fix it with patch release.

Prusa-Support, 2023-03-30:
This is a work in progress.
You can roughly monitor the progress at https://github.com/prusa3d/Prusa-Link/releases.
We can't provide an estimate for the final release and its implementation.
Please mind that Prusa Link is at an early development stage.

Michele Moramarco
Prusa Research

oubrecht-com, 2023-05-07:
Next good way is add API key as parameter into url. For example: http://192.168.1.200/?key=my_api_key
I have printer only in private network and it is not any secure risk for me.

MikeKovarik, 2023-05-07:
bump

this does not and should not take a year to make.

1. Add checkbox saying "Remember this API key for next time?" to the login popup
2. Store the api key to `localStorage`
3. Next time the app decides it needs to nag user for the api key, it first looks it up in `localStorage`, otherwise opens the annoying popup.

It's that simple, 10 lines of code at best. Here, I even wrote it down here for you. Just copy paste it. Though it's raw html & js, I don't know what framework you're using, I haven't looked into the code yet. But I can even create a PR for you if you won't shoot it down for security reasons.

in the popup inside `<form>` add:
```html
<input type="checkbox" name="remember" id="remember">
<label for="remember">Remember this API key for next time?</label>
```
and in the submit handler add
```js
const remember = document.queryString('input#remember').checked
if (remember) localStorage.apiKey = theVariableWithApiKeyGoesHere
```
and then wherever needed in intialization
```js
const {apiKey} = localStorage // load from localStorage to prevent multiple sync access' to it
if (apiKey && apiKey.length > 0 && /* some validity check for sure */) {
  // don't annoy user here
} else {
  // open the popup
}
```

Sure this is just per-device and even per-browser partial solution. And some undoubtedly see (honestly minimal) security threats in this solution. I don't disagree. But we're all hobbyists here. Let's not be so uptight, and just add some security warning there saying don't remember the key on public computers (common sense) and we're done here. At least until safer solution can be properly developed.

oubrecht-com, 2023-05-07:
Hi, I use for autologin to PrusaLink Firefox with extension (AddOn) "Header Editor". I make there rule for specific IP address that this extensinon has change "x-api-key" in HEAD.

Extension for Firefox is here: [here: https://addons.mozilla.org/en-US/firefox/addon/header-editor/](url)

Here is printscreen of settings of this my extension and exportet my rule:

Export:
[HE_2023-05-07T09-04-04Z.json.zip](https://github.com/prusa3d/Prusa-Link-Web/files/11414390/HE_2023-05-07T09-04-04Z.json.zip)

Printscreen:
[image]

MikeKovarik, 2023-05-07:
That's a good idea. But I'm mostly using the prusalink from my phone to check if I should get off the couch yet 🫤

mix579, 2023-05-07:
@oubrecht-com That's actually a great stopgap solution until a proper mechanism gets implemented. The same extension is available for Chrome, which I'm using. Works perfectly.

Prusa-Support, 2023-06-01:
I'm happy to see you sharing ideas and alternative solutions but rest assured, we are working on this.
On a side note, we don't really consider making the API visible in the URL a viable solution - https://github.com/prusa3d/Prusa-Link-Web/issues/325#issuecomment-1410202509. Even when you are connected to the local network, adding an authentication value in URL would be less safe than what we aim for. 

We don't have a timeframe yet, but the solution will most likely come along with the next "big" MINI firmware upgrade (4.5.x or 5.x.x). Also, the MINI in the future will be available for Prusa Connect, allowing safe remote connection even outside of the local network.

A lot of propaedeutic job has been already done and I seem to understand there is still quite a work ahead to be done.
Thanks for bearing with us.

Michele Moramarco
Prusa Research

mix579, 2023-06-01:
> the MINI in the future will be available for Prusa Connect

This day can't arrive too soon!

pabutterworth, 2023-06-09:
The Mini does seem to be the forgotten cousin, firmware enhancements are so slow

bolsoncerrado, 2023-07-27:
+1 for the MK4 fix.

Also explained politely here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3161

fhollermayer, 2023-08-12:
If I'm not mistaken this is still an issue – and I really don't get why. 

Sure, new features are in focus.  An updated authentication strategy that renders the API keys approach outdated is reason enough to reduce work on stuff like this issue to an absolute minimum.  I get that switching over to `localStorage` is a security concern and implementing cookies goes against the switch to username/password auth. But in it's current state  – one **year** after this issue was created – Prusa Link Web for the MINI+ is pretty much only usable to upload files and start prints via PrusaSlicer. There was time spent developing a web application that nobody uses because it requires an API key to be entered every single session.

And all that while the fix to the initial problem would be to replace the string `text` with `password` in `templates/components/modal/apiKey.html`

MR2C280, 2023-09-17:
For those of us who are not in a production enviroment (home user), security is of ZERO importance.  Who/why would anyone want to hack into my printer?  Can we please just have a option on the printer that turns off Prusa Link security altogether?  (Prusa Connect is another matter).

joelspadin, 2023-09-24:
Switching to HTTP digest authentication is actually worse for me. Previously, I could work around this by telling my password manager (Dashlane) that the API key field should be filled with a username and saving my API key as the username. Now, on my MK4, there is no such option because Dashlane will only auto fill digest authentication for HTTPS.

LMDavid, 2023-09-25:
This is what i've come up with using TamperMonkey until Prusa releases the new final firmware so i can use the browser password manager:

```javascript
// ==UserScript==
// @name         Prusa Link - Make API field a password type field
// @namespace    http://tampermonkey.net/
// @version      1.0
// @description  Changes the API field type to 'password' so the browser can remember the API key
// @author       LMDavid
// @match        http://prusa-mini/
// @grant        none
// ==/UserScript==

'use strict';

(new MutationObserver(check)).observe(document, {childList: true, subtree: true});

function check(changes, observer) {
    let el = document.querySelector('#apiKey');
    if (el) {
        el.type = 'password';

        let parentEl = el.parentElement;
        let bt = parentEl.querySelector('button');
        bt.addEventListener('click', () => {
            observer.observe(document, {childList: true, subtree: true});
        });

        let formWrapper = document.createElement('form');
        let frag = document.createDocumentFragment();
        while (parentEl.firstChild) {
            frag.appendChild(parentEl.removeChild(parentEl.firstChild));
        }
        formWrapper.appendChild(frag);
        parentEl.appendChild(formWrapper);

        observer.disconnect();
    }
}
```

You should change `http://prusa-mini/` with your IP address.

Burtfaceman, 2023-10-07:
@LMDavid , your script works great! Thanks!

For everyone who doesn't know (like me up until today): [Tampermonkey](https://www.tampermonkey.net/) is a browser extension that alters the execution of a web page using scripts stored locally on your device. Install the plugin, add LMDavid's script (modify as instructed), and your browser will save the API key as if it were a password.