prusa3d/Prusa-Link-Web #224: [BUG] PrusaLink doesn't like filenames with a space in them

kellertk, 2022-08-11:
**Printer type** - Prusa Original MINI+

**Printer firmware version** - 4.4.0-BETA2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
USB (PrusaLink)

**Describe the bug**
The Web UI will get 404s if you try and upload a file with a space in it and print the file. The print actually still happens, but there's a 404 when it tries to load the gcode file for `progress-with-img`.

For example:
GET `http://10.11.0.33/api/files/usb/Filament guide_Wheel_0.6n_0.15mm_PETG_MINI_12m.gcode` -> 404'd!

But if you try a new print without the spaces from the file name:
GET `http://10.11.0.33/api/files/usb/Filamentguide_Wheel_0.6n_0.15mm_PETG_MINI_12m.gcode` -> 200 OK

**How to reproduce**
1. Upload a file with a space in it.
2. Try and print the file. Printer starts, but UI shows a 404. The print status never loads in the browser.

**G-code**
All files with a space.

**Crash dump file**
N/A

Prusa-Support, 2022-08-17:
Thank you for the report.

It seems the file will load, display the preview, and print correctly if you launch the print later again.
Also, the bug is **not** related to a incompatible character I would say.
The bug occurs **only if** such print file is uploaded with **the option 'start print after transfer' enabled**.

Feel free to provide further feedback.
I will bring this to the attention of the developers.

Michele Moramarco
Prusa Research

machacekch, 2024-08-20:
prusa3d/Prusa-Firmware-Buddy#3039 was incorrectly closed. It's not a duplicate of this issue, but a separate one. As demonstrated in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3039#issuecomment-2298356730 that issue really is about file name length, not spaces or other special characters.