prusa3d/Prusa-Link-Web #232: [FEATURE REQUEST] Allow file rewrite for PrusaLink uploads

jkavalik, 2022-08-17:
When I try to reupload a file that already exists on the flashdrive, I currently get an error (403 iirc?). PrusaLink on MK3+RPi does not do that and rewrites the file instead. I find that very useful because it allows me to restart prints (after first-layer problems etc.) by just re-"upload and print" directly from PrusaSlicer instead of opening the PrusaLink in a browser and searching for the right file to start.

So would it be please possible to implement that on the Mini too?
- automatically rewrite already existing file if a file with the same name is uploaded.

ondratu, 2022-10-19:
We plan to do that. That is about API improvement first. So stay tuned, that will comes.

stefanct, 2023-03-25:
There has been a lot work done on the PL API in the last months. If this feature still require some more work maybe it would make sense to default to overwrite it (without choice) for now. That should be rather trivial to implement and an improvement for pretty much all users (I seriously doubt anybody relies on the printer to not overwrite gcode exports, at least I'd hope so ;)

DaReaLDeviL, 2023-04-17:
Please change that like requestet. I whould like that also and it is not the best way to always login in the printer. Another thumps up for this feature!

PR #393 (2023-05-29): Add confirmation modal to overwrite a file by upload
[image]

Closes #232
closed 2023-05-29 by Lowentwickler