prusa3d/Prusa-Link-Web #251: [Feature Request] Delete all gcode in my USB

dikarel, 2022-09-08:
For rapid prototyping, I use the "slice and print" option very frequently on PrusaSlicer. As a result, my USB on the 3D printer is full of gcode files that are generically named (e.g. something like `Body1_0.4mm_PLA_MINI_15m.gcode`). This creates 2 friction points:
- When I slice and print, I would occasionally run into a file-naming conflict with a past gcode that just happens to have the same auto-generated name. When this happens, it's quite frustrating because I have to wait for the (slow) file upload before I get the error message
- My USB is full of past prototype prints and looks disorganized

I wish there was a "delete all gcode" that I can easily trigger from time to time to clear up gcodes from past prints that I no longer want, so that I can avoid file naming conflicts. Today, I have to delete old gcode files one-by-one using a mouse pointer -- this feels inefficient.

Some alternative solutions:
- Using checkboxes, select specific gcode files and then bulk-delete
- An option to auto-archive or auto-delete gcode files older than X number of hours
- Allow deleting individual files using keyboard shortcuts (so the user can do it much faster without needing to use a mouse)

dikarel, 2022-09-08:
**Related question:** Would the maintainers appreciate code contributions for this? I wanted to double-check that this is in alignment with the product vision before investing effort

ondratu, 2022-09-08:
Thank you for your well described feature request!

At this moment I'm not sure about main **Delete All** functionality at this moment. May be, later, we really implement something like that. But your Alternative solutions are what we plan to.

But as you understand, most of that is about backend functionality, which have to be implemented on each type of our printer backends, instead of the last one. So i'll add keyboard shortcuts to our issue list.

About your relation question.
It is little bit more complex to answer, but simple yes, we are really appreciate community code contributions. But your requests are more about product directions and UI concept. So we have to figure out that first.

Prusa-Support, 2022-09-23:
This probably concerns both, Prusa Link and MINI firmware, but in the latter repository, we have a similar (or specular) open issue.
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2447

These issues won't necessarily create a conflict between duplicates so we may keep both open for now.

Michele Moramarco
Prusa Research

vanboom, 2023-08-09:
For your consideration, an AJAX type handling of the delete action would allow the UI to stay on-screen without refreshing after each file delete.  It is the refresh that causes the users to have to scroll down to find files again when deleting multiple files.