prusa3d/Prusa-Link-Web #255: [Feature Request] Prusalink Requirements for MMU2

jeremyfaller, 2022-09-12:
Is there a list of requirements that are needed to make Prusalink work with the MMU2? I might have time to jump into adding the features, but I need a little guidance on what's missing.

Thanks in advance.

ondratu, 2022-09-13:
At this moment, MMU is not supported. That means, that stable FW with stable MMU2 works, but Link ignore it.

We will work on MMU2 support with new FW for MMU and Printer, which is at this moment in early beta release.

jeremyfaller, 2022-09-13:
Are there things I can do to help?

ondratu, 2022-09-20:
Not yet, Thanks a lot!

Prusa-Support, 2022-09-23:
Thank you for reaching us.

It seems to be a matter of time. Prusa Link / Prusa Connect will be available for most of our products and users (arguably) soon!
We are working on this. Meantime, if all your questions in these regards have been answered, please consider closing this issue.
I'd leave this up to you but please mind that the issue can be opened again in the future if needed.

Michele Moramarco
Prusa Research

bearclaws8, 2023-10-31:
Do we need a separate issue to track MMU3 capability or can we use this issue to track both?  If we can use this issue to track both, can/should we update the issue title for clarity?  Having no one assigned to this issue after 12 months leaves me a little worried.  In traditional GitHub fashion, should this issue remain open until fully resolved?  Assuming that there is a plan to resolve...

@ondratu Can you clarify what doesn't work with PrusaLink and an MMU?  Will PrusaLink refuse to connect to a printer with an MMU?  Or is it something as simple as PrusaLink will not reflect any of the filament changes but that a printer with an MMU will still print gcode sliced for an MMU fine?  Would PrusaLink work with a printer an MMU if trying to print in "single" mode?

Brandoskey, 2026-04-27:
After 2+ years, is there any chance of MMU2 support in PrusaLink?