prusa3d/Prusa-Link-Web #258: [ENHANCEMENT] Set printer ready from the printer itself

darraghbr, 2022-09-21:
**Printer type** - MK3 but will also apply to MINI when that gets set up with Prusa Connect

**Is your enhancement related to a problem? Please describe.**
  I really like the 'print queue' function in Prusa Connect, where you queue up all the prints you need and once a print is finished and you have cleared the bed you can hit the 'Printer Ready' button in the web UI and the next print in the queue will start. 

What I would like to see is the ability to set 'Printer Ready' from the printer itself rather than having to go back to the web UI to do it. 

This could be with a simple prompt at the end of a print with just a 'Is the bed clear' Yes or No type question but ideally it would show what the next filename in the queue is, and what type of filament is needed. It could even guide you through a filament change if one is required. 

Clearly this enhancement would require a lot of work across both Prusa Connect and the firmware itself so I can put a corresponding issue into the Prusa-Link repo if required.

darraghbr, 2022-10-05:
This looks to have been solved in the most recent release (0.7.0-RC1) so should this issue be closed? Or is the current implementation a stopgap measure and this issue should remain for the moment?

Prusa-Support, 2022-10-12:
Thank you for reporting back.

The release [Prusa-Link 0.7.0.rc1](https://github.com/prusa3d/Prusa-Link/releases/tag/0.7.0.rc1) is actually a release candidate, which means it is most likely a definitive form, or very close to the stable release.

This said, it is safe to close the issue now, and it can be opened again later if needed; or, in case of further similar but more specific requests, you can open new issues in the future.

Michele Moramarco
Prusa Research

closed 2022-10-12 by Prusa-Support

m42e, 2023-09-13:
Dear @Prusa-Support would you please consider that for the MK4 ?

ssambdar, 2023-09-24:
Dear @Prusa-Support, I would love to see this feature in the future. I fill up my print queue for the day via Prusa Connect. When a print is done, I clean up the printer and right now I need to open Prusa Connect again only to start the next job. Why not just use a button on the MK4 to set the printer ready and start the next print?

darraghbr, 2023-09-24:
I think it's coming to the MK4 soon, it's just been integrated on the most recent MINI firmware as gets talked about here:

[Prusa Buddy Firmware issue number 3291](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3291)

Have you tried using the MK3 method where you 'print' a little g-code script on the machine?