prusa3d/Prusa-Link-Web #268: [Bug] Uploading gcode while one is printing changes the displayed file name

boydj, 2022-10-12:
Steps to recreate:

1) Start a print
2) While print is processing, upload another G-CODE
3) Observe that the name of the G-CODE changes on the Dashboard currently being printed, wrong name persists after a browser refresh

Using firmware 4.4.0-RC1 on Mini, with whatever prusa-link-web is bundled.

Prusa-Support, 2022-10-20:
Thank you for bringing this to our attention.

This obviously looks like a bug and will be labeled as that so that our developers can keep track of this and similar bugs and look into these in the future.

That's surely interesting but, for now, please mind that the tested scenarios for Prusa Link and MINI printer to cooperate correctly are described in the previous (beta) release note, and still apply.
>MINI must be on the home screen, “print-preview” screen, “print-finished” screen - this is intentional for safety reasons

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2

Also, this may look like a concern for Buddy Firmware [prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy). Not sure though, so I'd let it stay in this repository for now.

Michele Moramarco
Prusa Research

vanboom, 2022-12-08:
Writing to confirm that this is happening in the 4.4.0 production release.  The web interface displays the file name of the most recently uploaded file instead of the file that is actually printing.  Thanks!

eWuPi, 2023-03-02:
I confirm this error, not only does it display the last uploaded file, but when selecting reprint, it selects this file instead of the last one printed. In addition, it does not check the type of filament and, for example, after selecting "Print again", it started printing a PETG file on PLA. ...
---------------------
Potwierdzam ten błąd, nie tylko wyświetla ostatni wgrany plik, ale po wybraniu ponownego wydruku wybiera ten plik zamiast ostatnio drukowanego. Na dodatek nie sprawdza typu filamentu i np. u mnie po wybraniu "Drukuj ponownie" zaczął drukować plik PETG na PLA. ...

Prusa-Support, 2023-03-25:
This is a known issue at this point and will be taken care of over the next few releases.

Also, about the filament type check, that is an unrelated issue that will be most likely addressed together with issue https://github.com/prusa3d/Prusa-Link-Web/issues/281.

Michele Moramarco
Prusa Research