prusa3d/Prusa-Link-Web #271: Filenames with % character not valid

mix579, 2022-10-20:
Pardon me if I'm missing something obvious as I'm new to PrusaLink/PrusaConnect. File output is set to 
[input_filename_base]_{printer_model}_{filament_type[0]}_[temperature]C_[layer_height]mm_[fill_density]_[print_time].gcode
in all my profiles. When [fill_density] is filled in, it results in adding something like "_10%" to the filename. I never had an issue with that while using Octoprint but when I try to send that file from Prusaslicer or upload to PrusaConnect, I get an "invalid file name" error. I can correct it by deleting the % character. But what's particularly annoying is that the percent character is actually added in the first place by the slicer itself, not by me. Can the % character be added to the list of allowed characters?

mix579, 2022-10-29:
Just to add this to my comment, this is NOT an issue with PrusaLink on the Mini! It accepts the % character, whether send from Prusaslicer or by file upload.

Prusa-Support, 2022-11-05:
Thank you for reporting this.
There is a number of special characters that may cause unintended behaviors on the printer or on the Prusa Link side.
Sometimes they are quite uncommon characters, and sometimes characters are rather normally used.
We are taking notes, thank you again.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-11:
Here is an update.
We are putting together a list of "forbidden characters" that may be a cause of anomalies and incompatibilities in Prusa Link if included in _file name_ or _path_.
So far the listed characters are:
`\\`, `?`, `"`, `%`, `¯`, `°`, `#`, `ˇ`.
This list may grow or be modified in the future as our developers identify more problematic characters or act to improve Prusa Link in these regards.

These characters may have a specific meaning for Prusa Link and therefore they should be avoided to prevent problems for now.
We are still collecting information and your feedback is most valuable.

Michele Moramarco
Prusa Research

mix579, 2022-11-11:
That's fine to have a list of excluded characters but
a) it should be consistent across Mk3S and Mini implementations of PrusaLink
b) output variables in Prusaslicer should be modified to NOT include such characters (I know infill percentage does, and I suspect others do, too, maybe replace the character % with the letters pct).

Prusa-Support, 2022-11-24:
Sure. The issue will stay open for now.

Michele Moramarco
Prusa Research

Prusa-Support, 2025-08-19:
I seem to understand that Prusa Slicer no longer exports `%` character in the filename output, at least not in case of `{fill_density}`, rendering this issue potentially no longer valid. 

I suppose that's the optimal workaround because this character can't be accepted without causing disruption in the "file's URL" on Prusa Link/Connect, as quickly mentioned above, but I'm not sure if there are other placeholders and conditions that may automatically export `%` character at the moment.

Michele Moramarco
Prusa Research