prusa3d/Prusa-Link-Web #272: [FEATURE REQUEST] Order the Storage file listing by most recent date at the top

vanboom, 2022-10-23:
Suggestion: reversing the storage file listing order to show the most recent gcode files at the top would be a great workflow improvement vs. having to scroll to the bottom to find the newly uploaded files.   Thanks for a great software - I will attempt a PR if I can get a dev env set up for this.

ondratu, 2022-10-25:
Thanks, for your suggestion. Yes, we plan to sort files over date and name ascending and descending, as user want.

sbierly, 2023-04-22:
Seconding this *old* feature request. It would be a trivial fix to sort like the printers do, most recent first, even if that's the ONLY thing you do. Adding fields and customizable sort sounds great, but let's not wait for elegant solutions, just please break this broken workflow.

Prusa-Support, 2023-05-17:
@sbierly same as the other issue - https://github.com/prusa3d/Prusa-Link-Web/issues/274#issuecomment-1518748647.

There is a lot of work behind every single release and announced project.

We are aware of the current limitations so please bear with us while we are working on this.
For open discussions, please pay a visit to our [forum](https://forum.prusa3d.com/).

Michele Moramarco
Prusa Research

PhilipWhiteside, 2024-12-30:
I made a browser script to do this https://gist.github.com/PhilipWhiteside/8a60bcc6d48a91d50b1bae9e7120f8fa.

danopernis, 2024-12-30:
Created https://github.com/prusa3d/Prusa-Link-Web/pull/515 as soon as this gets merged I will be happy to update static Prusa Link assets in Buddy Firmware. There should really be no need to use browser scripts, especially when this is already implemented and only needs to be enabled at this point.

commit 709dd11d (2025-01-13): mini+friends: Enable sorting by date and size

Resolve #272
closed 2025-01-13 by ondratu