prusa3d/Prusa-Link-Web #276: [BUG] PrusaLink ERR_CONNECTION_REFUSED

ThomasCZ, 2022-10-31:


Prusa-Support, 2022-11-02:
Thank you for reporting.

There is a chance that there was an intentional temporary interruption of the service on our side. Not sure though.
Does the issue persist?

Michele Moramarco
Prusa Research

ThomasCZ, 2022-11-02:
Yes, I am able to access the PrusaLink only after every WiFi module firmware flash. When I turn off and on the printer, I am getting always the `ERR_CONNECTION_REFUSED` message.

Prusa-Support, 2022-11-02:
So the error persists but it doesn't occur always, right?
Did you notice a change of IP in the network settings of the printer?
Does the problem persist if the printer is connected via Ethernet cable?

Have you tried a factory reset, a new firmware installation, and a new ESP module configuration (Wi-Fi  Wizard or Setup Wi-Fi)?

What are the exact steps to reproduce the issue?

Michele Moramarco
Prusa Research

Eldenroot, 2022-11-02:
Only happens with wifi module. 

I have tried fixed and static IP, still same.

ThomasCZ, 2022-11-02:
These steps are always reproducible in my case
1. Network -> WiFi -> Set WiFi module
2. Open IP in browser and set API key -> Success
3. Turn off the printer
4. Turn on the printer
5. Wait
6. Open IP in browser -> `ERR_CONNECTION_REFUSED`

ThomasCZ, 2022-11-02:
The printer is showing completion time, not remaining time, so the printer is online and knows current time. But I can not connect to PrusaLink.

Eldenroot, 2022-11-02:
> The printer is showing completion time, not remaining time, so the printer is online and knows current time. But I can not connect to PrusaLink.

Exactly, same thing here.

Prusa-Support, 2022-11-05:
I'm sorry, I am still unable to reproduce this.

Let's please introduce check steps after the n°6.

6. Open IP in browser -> in case of "ERR_CONNECTION_REFUSED"...
7. Check if the WiFi icon on the printer's display is greyed or bright white
8. Check in Network Settings if the IP is now different than the address on the browser

Also...
>The printer is showing completion time, not remaining time, so the printer is online and knows current time. But I can not connect to PrusaLink.

Does that mean there is an additional step where you start a print?

Michele Moramarco
Prusa Research

Eldenroot, 2022-11-05:
Obviously  he is not the only one with this error, because I can reproduce here as well.

For the second part - I annot connect to the printer (PrusaLink), even when I turn off/on the printer or restart the PC... but the printer can get the time and date from the internet... so it means that the ESP is working and communicate.

Only PrusaLink is not working... so this is the evidence that it is not about some blockage or network issue...

ThomasCZ, 2022-11-07:
I tried to flash WiFi module firmware again and I am still not able to connect to the PrusaLink. Wifi icon is white and the printer is showing print completion time, not remaining.

[dump.zip](https://github.com/prusa3d/Prusa-Link-Web/files/9949181/dump.zip)

ThomasCZ, 2022-11-08:
Two WiFi modules ordered from [Aliexpress](https://www.aliexpress.com/item/32809618395.html?spm=a2g0o.order_list.0.0.31211802v7PAip) and two printers with different Buddy board version. Version 1.0.7 is OK, mine (1.0.5) is not working. Unfortunately, I cannot try my module on 1.0.7 a vice versa.

I can send you (Prague) my module and you can try to solve the issue.

**NOK**
[image]

**OK**
[image]

Eldenroot, 2022-11-08:
Yes. Mine is 1.0.5

Prusa-Support, 2022-11-11:
Have you experienced issues with [FW 4.4.0-beta2 ](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2)? Have you tried downgrading?

.

The redundant data about problems with rev 1.0.5 is surely interesting and probably a hint.
Funnily enough, though, I personally had the chance to test a Buddy 2020 rev 1.0.5 and I've not been able to reproduce the issue.

>Two WiFi modules ordered from [Aliexpress](https://www.aliexpress.com/item/32809618395.html?spm=a2g0o.order_list.0.0.31211802v7PAip) and two printers with different Buddy board version. Version 1.0.7 is OK, mine (1.0.5) is not working. Unfortunately, I cannot try my module on 1.0.7 and vice versa.

@ThomasCZ. Not sure I understand what prevents you from swapping ESP modules between the printers.

I think the ESP module on the link is compatible and was listed as "community approved" in the linked sheet under [_Hardware prerequisites (4.4.0-beta2 notes)_](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2), but please double-check the data. As of now, it is listed on _line 51_, and it says 1MB cache while the product page doesn't specify this - 1MB cache recommended.

>The ESP8266 is quite popular among other devices, especially in IoT (internet of things), therefore there are multiple manufacturers and also two hardware revisions. We have purchased several Wi-Fi modules from different producers and our internal tests haven't shown any significant differences, however, as with any other product, not all might be equal. For this reason, we are sharing the data from our [measurements in this spreadsheet](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit?usp=sharing) and we would appreciate it if the community would join us.
>
>Regarding hardware revisions, there are two, ESP-01 and ESP-01S. Both differ in memory size (512 kB vs 1 MB flash) and assembly of the LED. During the tests, we haven't discovered any functional differences between them. However, it is recommended to purchase the higher memory (1 MB flash).

@Eldenroot. What's yours?

Michele Moramarco
Prusa Research

ThomasCZ, 2022-11-15:
There is around 9 600 km between these two printers. I have ESP-01S and the printer is able to download current timestamp and I have been able to access PrusaLink twice, so I think, the module is OK. Anyway, It looks like on never ending ping pong, so I ordered another "community approved" module.

ThomasCZ, 2022-11-25:
I have new WiFi module from another recommended seller and there is the exactly same behavior as before. I can access PrusaLink only for the first time after flash WiFi module firmware. When I type API key and just close the browser tab and open it again, there is `ERR_CONNECTION_REFUSED` error. Same behavior also on new 4.4.0 firmware.

ThomasCZ, 2022-11-25:
I tried mobile hotspot. It works only for the first time, close and reopen browser tab and PrusaLink was dead. I tried to load hotspot settings many times but Prusa Link was inaccessible. I loaded my home WiFi settings many times, because of gray WiFi icon event after restart the printer. After many attempts, the WiFi icon was white and PrusaLink is accessible even after close browser tab, restart the printer by switch or by HW button under the LCD.

Many times the printer shows me "WiFi module flash screen" by itself.

Prusa-Support, 2022-11-29:
Please make sure not to confuse the known issue '_Wi-Fi (ESP) ignoring printer restart_' described in the log notes of the [final FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0).
As of now, the Wi-Fi module may not reboot correctly after a short power cycle.  Please power OFF the printer and power it back ON after letting the printer rest for a few seconds.

Maybe test a different network - the mobile hotspot could be a good idea too - but be aware of the known issue.

Michele Moramarco
Prusa Research

OJFord, 2023-06-20:
I have the same issue with firmware 4.4.1 final. Also Buddy board 1.0.5.

Prusa-Support, 2023-08-25:
Power cycling may help if the ESP Wi-Fi module bricks for some reason.

After a power cycle, please wait several seconds for the ESP Wi-Fi module to boot up and establish the Wi-Fi connection correctly. 
Please wait for the Wi-Fi icon to turn up bright in the upper part of the display.

Michele Moramarco
Prusa Reseach

OJFord, 2023-08-26:
Minutes or hours, it doesn't make a difference. As others have described it does show the icon indicating it has connected successfully, it does report a DHCP-acquired IP address, it just doesn't respond to pings or loading the dashboard in the browser.