prusa3d/Prusa-Link-Web #309: [FEATURE REQUEST] Storage navigation, sort and selection

Danbuhr, 2022-11-30:
I would like to have sort and edit handling of storage files similiar to that of Octoprint.

Sort by name, date of upload, filesize etc.
Select files/folders - all or some and delete.

ondratu, 2022-11-30:
Hi, thanks for your suggest. We have implemented sorting  yet. That will be usable in new MINI release and PrusaLink for MK3 release.

nrnrnr, 2023-10-04:
Would love to have this for the MK4 as well.  Particularly the ability to sort files by date, as they appear on the printer's screen.

meermanr, 2024-01-02:
FYI, Prusa-Firmware-Buddy v5.1.0 (on my XL5T) still sorts alphabetically, and I cannot find any controls to change it to by date.

trinxie, 2024-01-02:
I agree as well. Have moved from Octoprint to PrusaLink, currently only miss sorting by date (latest) + the camera integration.

ChrisHeerschap, 2024-02-11:
5.1.2 and still nothing. Gotta figure other things are taking priority because changing the sort order doesn't _seem_ like a particularly challenging update. Maybe it's the UI change for selecting sort order, either way, this is yet another way Prusa Link is a long way from even coming close to replacing Octoprint.