prusa3d/Prusa-Link-Web #325: v4.4.0 makes automated logging of printer telemetry impossible

twshield, 2022-12-16:
Version 4.3.x had a very useful URL of 

http://[prusa mini hostname]/api/telemetry

that returned a simple json file that allowed logging of the printer status (temperatures, file printing, etc).  I found this feature so useful, I'm going to revert my firmware until this is restored.

With the network changes in v4.4.0 and the need for a API key for every session (which also makes using the web interface from a browser annoying), I don't see any way to get the equivalent information from the new web interface.  I see no reason the read only information in the v4.3 json telemetry needs to be protected by the API key authentication mechanism.

Furthermore, the printer won't even ping, so there is not even an easy way to tell if the printer is up with the new firmware.

ondratu, 2023-01-04:
Hi, yes, we are working on new API, which is unfortunately not complete yet.

The new way is authorize any resources by HTTP Digest (API-Key for your firmware version). So there will be no way, how to get any info without knowing that key and password in the future.

The new API is describe here: https://github.com/prusa3d/Prusa-Link-Web/blob/master/spec/openapi.yaml

We will discuss, if and how to allow simple ping if printer is alive. Probably some HEAD request on /api/version

twshield, 2023-01-04:
Thanks for the reply.

Needing an API key/password is fine if there is a way to specify them in the URL, such as

http://[prusa mini hostname]/api/telemetry?apikey=ABCD1234

You can't just turn ping responses on in the network layer?  I would think you are making support more difficult for yourself by not allowing users to check network connectivity with ping.  It's always the first thing I do to check when there is a network problem.

Prusa-Support, 2023-01-31:
Thanks for your suggestions.
I guess making the API key/password "explicit" in the URL would render the printer vulnerable.
I wonder if a feature to enable/disables the protection key could be an option at this point- at least when connecting within a local network - but safe connecting is an important matter to us.

Our developers will make their considerations about that.

In near future, the browser should be allowed to autofill credentials (https://github.com/prusa3d/Prusa-Link-Web/issues/222) and this could shorten the authentication process at least.

Michele Moramarco
Prusa Research

LucaMH, 2023-01-31:
Idea: add a checkbox in the settings to allow passwordless/keyless reading from the api and let it disabled by default

twshield, 2023-01-31:
That seems like a good solution. Thanks.

Prusa-Support, 2023-02-14:
Thanks for sharing your ideas.
We are taking notes.

Michele Moramarco
Prusa Research