prusa3d/Prusa-Link-Web #328: Prusa SL1S - Prusa Link Web, Why is the build plate preview so small?

Area5142, 2022-12-24:
Connecting to Prusa SL1S Prusa Link with the browser (Firefox and Google Chrome) gives me this Dashboard when printing:
[image]

The build plate preview is so small and there seems to be plenty of room for a bigger picture/image. Why is it so small?

I tested with both newest Firefox and Google Chrome on Linux Mint 19.3 x64 system.

Lowentwickler, 2022-12-24:
I believe we've changed this behavior. However SLA printers are not using the latest version of the web UI. I will check it next week and let you know.

Prusa-Support, 2023-01-19:
Thanks for reporting this.

This should be fixed on the printer's firmware side.
I'll transfer the issue to the relevant repository now.

Michele Moramarco
Prusa Research