prusa3d/Prusa-Link-Web #330: Can't cancel transfer

mix579, 2023-01-01:
File transfers from Prusaslicer sometimes get stuck. Config: Raspi 3b on a Mk3S+ printer, connected by USB. 

For example, printer is turned off but the Pi is on when the file is sent from Prusaslicer. 

Screenshot: https://i.imgur.com/vcx5Xgr.png

The screen says Done but the file never makes it to the printer. The transfer never recovers and blocks all other uploads. In fact I found one on one of my printers that apparently had been sitting in that state for more than a day. The only way to fix it that I found is to reboot the Pi.

Prusa-Support, 2023-02-21:
Thanks for your precious feedback.

We mostly test and recommend Raspberry Pi Zero W which would be powered off together with the printer, rendering this problem less likely to occur.
It would surely be interesting hearing about the users experience with other boards. We are taking notes.

Michele Moramarco
Prusa Research

sbierly, 2023-04-22:
A cancel button would be most awesome, particularly given how painfully slow transfer on the Mini internal Wifi is. Currently, you have to walk to the printer and reboot it, risking something indeterminate happening.

Prusa-Support, 2023-05-26:
A remote reset button should actually solve this issue.
This feature is newly available via Prusa Connect - an alternative option may also be available in Prusa Link in the future (unconfirmed).

This issue may be solved. Please let us know if the 'reset' button in Prusa Connect helps.

Michele Moramarco
Prusa Research