prusa3d/Prusa-Link-Web #331: Feature Request: Issue a notification when a print completes

kevinoliver, 2023-01-03:
It would be helpful to have an OS-level notification when a print completes.

kevinoliver, 2023-01-03:
I'd be interested in contributing this if there is interest from the maintainers.

mix579, 2023-01-03:
Great idea!

Lowentwickler, 2023-01-04:
The feature is in our TODO list. Should be implemented soon enough.

ondratu, 2023-01-04:
This functionality is in Connect, click on Bell icon next to your login avatar. Link have to stay simple.

closed 2023-01-04 by ondratu

kevinoliver, 2023-01-04:
> This functionality is in Connect, click on Bell icon next to your login avatar. Link have to stay simple.

I'm not seeing this functionality in PrusaLink and this doesn't seem like it would make PrusaLink be "not simple". @ondratu can you explain a bit more?

Prusa-Support, 2023-02-01:
_Prusa Link_ is the most **basic interface layer** between the printer and _Prusa Connect_.
_Prusa Connect_ will be the ultimate **printers manager platform**, capable of advanced features.
https://connect.prusa3d.com

Michele Moramarco
Prusa Research

kevinoliver, 2023-02-01:
> _Prusa Link_ is the most **basic interface layer** between the printer and _Prusa Connect_. _Prusa Connect_ will be the ultimate **printers manager platform**, capable of advanced features. https://connect.prusa3d.com
> 
> Michele Moramarco Prusa Research

Michele Moramarco, thanks for explaining. 

I have a Mini+ and as far as I can tell _Prusa Connect_ does not support my printer. 😢

Prusa-Support, 2023-02-14:
You are welcome.
No, the MINI is not supported by Prusa Connect, not now, not yet.
It is a matter of time though.

Michele Moramarco
Prusa Research