prusa3d/Prusa-Link-Web #373: How can I use prusalink API ?

JeanHerbaut, 2023-03-12:
Hello,

Here is the project.

I want to use the PrusaLink API
Http://ip/api/job to retrieve the current state of my printers'job and send information by SMS to my mobile Phone depending on the progress.

I thought about doing and API request, setting my API key and credentials but the connection is always denied.

Does anyone had this problem or use the API properly ?

Thanks

Lowentwickler, 2023-04-25:
Hello!

- What kind of the printer model do you have?
- Have you tried DIGEST authentication?

joltcan, 2023-05-23:
I just got the mk4. To get the progress I figured out this:

```bash
curl -H "X-Api-Key: ${PRUSAPW}" -s http://mk4.hemma:80/api/job | jq .
{
  "state": "Printing",
  "job": {
    "estimatedPrintTime": 9295,
    "file": {
      "name": "HatPlate_0.4n_0.3mm_PLA_MK4_2h32m.gcode",
      "path": "/usb/HATPLA~1.GCO",
      "display": "HatPlate_0.4n_0.3mm_PLA_MK4_2h32m.gcode"
    }
  },
  "progress": {
    "printTimeLeft": 6840,
    "completion": 0.24,
    "printTime": 2455
  }
}
```

`${PRUSAPW}` is a variable I have in my script (`export PRUSAPW=<my pw from Network - Prusalink>`).
`jq` is a program that formats json.

joltcan, 2023-05-23:
And if you want to overcomplicate things you can use `jq` transform the progress into a number and do this in a shell script.

```bash
PRUSAPW="<my pw>"
if [ $(curl -H "X-Api-Key: ${PRUSAPW}" -s http://mk4.hemma:80/api/job | jq -r '.progress.completion|tonumber * 100') -eq 100 ]
then
    <some command to send sms>
fi
```