prusa3d/Prusa-Link-Web #392: Orange screen of death "malloc out of memory" error

sbierly, 2023-04-18:
The printer has been on for several days. One week old. Firmware up to date as of days ago. I did a drag and drop of a 5MB Gcode file, and the printer immediately began the error tone and orange screen. Accepted it's offer to reset, dragged again, and it is now printing. 

While I'm here--please find a more elegant secure login solution, having to manually enter the token every time is rather barbaric and confusing. Also we should be able to name each printer and clearly see that on each web page, currently you have to do an IP address correlation to make sure you have the right printer.

sbierly, 2023-04-28:
> Hi, please, could you tell us what kind of printer do you report that
> issue? And which firmware do you have on it.

Brand new Mini+, latest firmware  FIRMWARE 4.4.1
--Scott

sbierly, 2023-04-28:
Oh wow, by coincidence it just did it again (only the second time). This
time, completely different:

   - Mini+  FIRMWARE 4.4.1 has been up for days, perhaps even a week
   - I had used Prusalink to upload several times in that uptime without
   issue, other than the other bug I reported where you can't remote print due
   to not being at the machine home screen
   - I have also printed locally many times without incident, the thumb
   drive came and went several times
   - I tried to print, for the 3rd time in a row, a single small job with
   one small item (45 min print)
   - This time, immediately upon locally hitting "print", it triggered the
   orange screen, "malloc out of memory, click here to restart machine",
   something to that effect.

Nothing like this has ever happened before, so it seems like a firmware
bug, possibly due to 4.4.1, more likely due to using Prusalink
--Scott

sbierly, 2023-06-22:
This bug is getting annoying. I have yet a different case to report. This
time:

   - The machine had just been powered on
   - I started a known good job from SD card (the card was already inserted
   and untouched)
   - I watched it start and left
   - An hour later, I heard screeching from the basement, and investigating
   found the orange "malloc out of memory" screen again and constant tone
   - At this point I remembered that some time earlier I had gone to my
   Prusalink tab to check the progress of the print, and it would not load the
   web page
   - Hard to know what is cause and effect, but either nothing more than
   the installation and enabling of the wifi card can cause this, or a web
   connection to Prusalink can cause this, in the middle of a print with no
   other complications.

Very frustrating, wasted a couple hours and some expensive TPE filament.
Capabilities should not be released from Beta when they are in such a
shabby state.
--Scott

leon-matthews, 2023-07-19:
I've been hoping for a firmware update for a while as I get this same error every week or so on my Mini+, shipped mid-Jan 2023, running firmware version 4.4.1.

Doesn't seem to have a pattern to it, so I guess you've just got a memory leak somewhere that takes a few days of work to accumulate. My usage pattern might be a clue: I do use PrusaLink over LAN  to upload all of my models, and I keep the printer turned on 24/7. Hope this helps!