prusa3d/Prusa-Link-Web #398: "409 Conflict can't start print right now"

sbierly, 2023-04-22:
recurring occasional problem, sometimes things work, sometimes the file transfers, but it refuses to remotely start. Walk over to the printer and it prints just fine.

mix579, 2023-04-22:
That seems to be the state of affairs right now for all platforms that are not the old Mk3S/Raspi combo. I have not been able to start one print from my XL without having to walk over and press print on the preview, which totally defeats the purpose of remote printing. Same for the Mini, where probably half the time the print starts and the other half it sits on the preview screen, or worse, it just craps out because the printer LCD is not showing one of the menus Prusalink likes.

This has got to be fixed ASAP.

dpm0, 2023-04-25:
It is frustrating, when you just quickly try to start a job before going to bed. No No No, "Cannot start print at this time" 409

dpm0, 2023-04-25:
Just for interests sake, my error listed above was caused by the 100% Complete dialogue still showing on the screen from the previous job. Once I dismissed that and went back to the main menu on the printer, it worked as expected.

nvtkaszpir, 2023-04-26:
I'm seeing the same issue on my device.
Looks like after finished print it just shows on LCD 100% complete and waits for user action, which causes conflict when interacting via web interface.

Model Prusa Mini+
Firmware: 
API version ```{"api":"2.0.0","server":"2.1.2","text":"PrusaLink MINI","hostname":"PrusaMINI"}```

dpm0, 2023-04-26:
I imagine a fix will be given priority, once the hectic time of a new model dies down a bit. :)

dpm0, 2023-04-30:
Pressing the "reset" button on the printer's control panel after every time you remove the last completed print job from the printer, gets rid of this error. Not a fix, but a work around.

sbierly, 2023-04-30:
So, as some others figured out, this particular bug seems to happen when
you leave the printer (Mini, anyway) in a state other than the home menu
screen. Since I've made sure I always returned to there, I've never gotten
this error and it uploads and prints fine. HOWEVER, a second but seems to
be some sort of memory leak related to using PrusaLink on the Mini, which
can orange screen freeze up at any time after using PrusaLink. For THIS
reason, I've taken to rebooting often, definitely before a long print. So,
for now, this appears to be good practice for more than one reason.
--Scott

ChrisHeerschap, 2023-07-28:
Got the wifi for the Mini and thought it was cool that I didn't need Octoprint anymore, but still have to remember to dismiss the "print complete" when I clear the part or I run into this. Could imagine it might be possible to send some sort of gcode to dismiss that screen remotely but it would have to be something that could be done from PrusaLink UI.

sbierly, 2023-07-28:
It's not worth anybody's time to try to hack their way around buggy
software. Rather, Prusa should really support ALL their products, not just
the new ones. This issue has been festering the better part of a year, and
Mini owners have gone 6 months without a firmware upgrade...
--Scott

mix579, 2023-07-28:
This affects all Buddy-based printers, not just the "old" Mini but also the "new" Mk4 and XL. It's mind boggling that it hasn't been fixed.

nvtkaszpir, 2023-07-28:
I just added remotely managed electric socket and turn it off/on after print is done 😆

sbierly, 2023-07-28:
OMG
--Scott

nvtkaszpir, 2023-07-29:
well, it's the only way to fix it now without touching the printer, because there is no web API endpoint to restart the printer (or it does nothing or returns error, can't remember).

jahnroux, 2023-08-11:
How is this still an issue? I can't imagine it's a complicated fix - have a setting for letting the "complete" screen timeout back to home screen? Have prusalink or prusa slicer connect override the screen? Seems like the only solution is a wifi switch to power cycle the printer.

jahnroux, 2023-08-11:
This appears to be tracked here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2301

ChrisHeerschap, 2023-11-08:
Looks like it's been fixed in the 5.0 alpha firmware for the mini, at least I haven't been seeing it again.