prusa3d/Prusa-Link-Web #399: [ENHANCEMENT] Default File Sorting / Filtering in Prusa Link

markdesouza, 2023-04-26:
Enter what type of printer or upgrade the enhancement applies to.
Prusa Mini +
Prusa Link

**Is your suggested improvement related to an existing problem? Please describe.**
Under the storage tab the files are listed in alphabetical order and macos index/meta files (ie ._<filename> ) are shown

**Describe the expected functionality**
The files should be displayed in descending modification time order (similar to the actual printer) and macos index/meta files should be hidden

JAngerCAN, 2023-12-17:
Yes, please add this feature.

JAngerCAN, 2023-12-20:
I have a work around - add the date & time to the file output with placeholders - now files are sorted in date order in prusa link, descending date order, but still helpful I think. Just add place holders to the output file name. Placeholders listed below. Screen shot of my config. 

https://help.prusa3d.com/article/list-of-placeholders_205643

[image]

JAngerCAN, 2023-12-27:
The string I use for output filename is: {year}-{month}-{day}-{hour}h{minute}m-{input_filename_base}_0.4n_{layer_height}mm_{printing_filament_types}_{printer_model}_{print_time}.gcode

twshield, 2024-01-24:
I agree that descending modification time order sorting of the files would be good and this would match what the printer LCD interface shows (which is very helpful there).