prusa3d/Prusa-Link-Web #423: Cannot login to PrusaLink on Ipad

VH-fixer, 2023-07-05:
I can login to PrusaLink from a browser in Windows and ion my Android phone.  Edge and Chrome on Windows, Chrome on the Android.  I know how to login. 
I am using the correct login data. 

PrusaLink opens on Ipad when address is entered.   Login window opens.    Login information is entered.    After submitting the login data the login screen re-appears as if you did not enter the correct data:
Safari, User ID and password entries are now blank.  There is a message "Your password will be sent unencrypted"  shown in the login screen.  
Chrome remembers the login ID, but password is empty.
trying to login a second time by re-entering the data still does not work. 

I have tried 2 Ipads, using Safari and Chrome browsers.   

Ipad OS 16.5.1

Lowentwickler, 2023-07-07:
Hello,
could you please specify the printer model and its firmware version?

cozmo14047, 2023-07-18:
ive got this error printer is on 3.11 i know 3.12 is latest but thermal model protection stops me from updating ive got latest prusa link version on pi 4b its on any device running safari by the seems of it tried on iphone ipad imac

Bikedrama, 2023-07-20:
This also happens on safari on macOS.  In chrome on macOS all is well.

VH-fixer, 2023-08-06:
This is on MK4, FW 4.64, 4.7.0, 4.7.1 and still happens on FW 4.7.2

astralDenizen, 2023-09-09:
Can confirm on FW 5.0.0-RC

Claghorn, 2023-09-28:
Just finished my MK4 kit build, loaded firmware 5.0.0, setup wifi, but I can't login with any browser. I just keep getting the login window over and over again.

pyrho, 2023-10-04:
The cause has been found, and a fix provided over at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3287.
We just need to hear back from prusa now.