prusa3d/Prusa-Link-Web #436: [ENHANCEMENT] Wyze Cam Support for Prusa Link/Connect

chrishilbert, 2023-09-27:
Please integrate with the Wyze APIs for camera support.  A large portion of the community is using Wyze cams because they are cheap and high quality to monitor our Pursa printers.

https://support.wyze.com/hc/en-us/articles/16129834216731-Creating-an-API-Key

bjohns029, 2024-01-27:
This would be very useful!

Mr3Deee, 2024-02-04:
Agreed! It is somehow really sad to see that fiddling around with an additional phone or a Pi is needed to get this done on a 4000$ printer. Setting up something which is already in use (if not already built in) sounds like what I'd expect.

ErikV1000, 2024-02-14:
I seccond that. I have been using Wyze cams for the last few years to monitor my printers and in the last upgrade round (MK3S+/MMU3's and MK4) started using Prusa connect. It would be great to be able to connect my camera's into the same overview screen (without having to buy new ones or find a bunch of phones).

calvin-smith, 2024-02-29:
I also would like to see this added.  The current options for cameras is really limited.

drewgt2, 2024-02-29:
Me too - would love to integrate my Wyze cam onto dashboard

NoSQLKnowHow, 2024-03-05:
+1 to this

codemonkey2k5, 2024-04-03:
Yes please add wyze camera option!

andyylin, 2024-06-13:
Wyze support would be amazing.

m10d, 2024-06-27:
hopefully this could be resolved with https://github.com/prusa3d/Prusa-Link-Web/issues/496 (and if it can't, IE wyze cannot offer standard video stream that the likes of v4l can consume - then perhaps this issue should be a lower priority and possibly piggyback later off work supporting open standards).

m10d, 2024-06-27:
I looked i up - wyze DOES offer RTSP feeds via a different installable (from wyze) firmware.  It's possible wyze supports other standard/open formats that would be trivial & most appropriate to integrate

russdogg, 2024-09-03:
In regards to RTSP firmware support, Wyze has pulled their official firmware that had that option. You *can* still find those files online elsewhere (but not on Wyze official site). It looks like they will not be supporting this in the future.

m10d, 2024-09-03:
> In regards to RTSP firmware support, Wyze has pulled their official firmware that had that option. You _can_ still find those files online elsewhere (but not on Wyze official site). It looks like they will not be supporting this in the future.

Well that's unfortunate. 

While its understandable everyone would want "their"  solution (eg wyze cams, for owners of wyze gear) - this just emphasizes why open solutions merit support, as well as prioritization for integration.  

These are not going to decide its not in their "business interest" to allow customers to use a standard integration - ~walled garden wor the win!~

Unflexibel, 2024-09-05:
Their **could** be a workaround in regards to WebRTC/RTSP/RTMP/HLS for a Wyze Cam

I have a WyzeCam V4 and using Home Automation with 
https://www.home-assistant.io/integrations/prusalink with this add-on: https://github.com/mrlt8/docker-wyze-bridge

maybe useful, maybe not ¯\_( ツ )_/¯