prusa3d/Prusa-Link-Web #445: No Gcode thumbnails after update to bgcode on connect / link working

yasars, 2023-10-27:
[image]
### Description of the bug

after slicing i have no thumbnails ... latest beta prusa mk4 firmware , latest slicer alpha but no preview anymore...

when i upload to connect no thumbnails no preview...

when i upload to prusalink .. ans link upload self to connect i have thumbnails , and preview... its a connect bug

### Project file & How to reproduce

no thumbnails no preview

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

latest alpha

### Operating system

mac

### Printer model

mk4 latest beta

SparrowSX, 2024-02-29:
I had the same problem. The printer settings - firmware (expert mode) had incorrect data for G-code thumbnails.
The image type should no longer be PNG but QOI. 

[image]

yasars, 2024-02-29:
@SparrowSX I'm already using the stock settings for the Mk4, the problem is turn off your printer... and push a bgcode file into the cloud... no preview image is displayed, only when the printer is online and you have the file " "open" you will see a picture..