prusa3d/Prusa-Link-Web #448: Nozzle Diameter is reported incorrectly.

BrienKing, 2023-11-04:
It shows it as .40 mm when I have the .6 installed and the part printed was sliced with .6.

[image]

darraghbr, 2023-11-27:
Depending on what machine this is you need to set the nozzle diameter in the menu of the printer. I know on the MK3 it's in the 'HW setup' menu and on the MK4 I think it's in 'Settings - Hardware'.

BrienKing, 2023-11-27:
Yes, it was set in the menu.

BrienKing, 2023-11-27:
XL

xnk, 2023-12-06:
In my case PrusaLink seems to report the nozzle diameter of the first extruder (0.8mm) even when printing with any of the other four extruders (with 0.4mm nozzles) on the 5T XL.

Gefionious, 2023-12-13:
I have the same problem. I installed 0.4mm nozzles on my XL-2T (both tool heads), went into the menu on the printer and selected 0.4mm and PrusaLink still stubbornly reports I have 0.6mm nozzles installed.

Gefionious, 2023-12-18:
Quick update on my side, this issue continues even with the latest firmware on the XL (5.1.2?). This includes using the new calibration process to select the nozzle size of 0.4mm for both tools. PrusaLink still reports 0.6mm nozzles incorrectly on both toolheads.