prusa3d/Prusa-Link-Web #455: Open Dialog for gcode upload should include .bgcode

shadow651, 2023-11-24:
Currently the Prusa Link open dialog (when uploading gcode to the printer) filters to *.gcode. With the introduction of Binary gcode the filter should also include *.bgcode

I'm running firmware v5.1.0 on the Prusa XL 5 Tool version

lggomez, 2024-01-11:
By choosing `All Files (*.*)` in the dialog it works on current version, but I cannot see the bgcode extension in my mk4

This is weird because bgcode is in the list of extensions allowed for upload in the printer context:

https://github.com/prusa3d/Prusa-Link-Web/blob/master/src/printer/fdm/context.js#L59
https://github.com/prusa3d/Prusa-Link-Web/blob/master/config.mini.js#L7

This for the mini rather than the mk4, but it's the only profile that somewhat matches the model

iridris, 2024-01-11:
I think this line might be where bgcode is missing from:

https://github.com/prusa3d/Prusa-Link-Web/blob/master/webpack.config.js#L31

floretan, 2024-02-06:
It is now possible to drag and drop bgcode files, or to select "all files" in the native upload selection, but adding `.bgcode` to the `accept` attribute of the file upload field would make the files selectable right away.

vkorenev, 2024-03-04:
I have this issue on MK4 with 5.1.3 firmware:
[image]

beeb, 2024-03-08:
bgcode should be included in the default options I agree. I'm surprised this hasn't been fixed in several months, since it's a very easy change to make.

iridris, 2024-04-17:
This is fixed by the 6.0.0 release of the Buddy Firmware: 
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0

vkorenev, 2024-06-03:
I can confirm this is fixed in 6.0.0. This issue can be closed.