prusa3d/Prusa-Link-Web #465: Authentication prompt window does not work with password managers

MisterGlass, 2023-12-15:
The current prompt implementation does not work well with password managers, leading to a poor user experience. A more traditional HTML form would allow password managers and other tools to interact with the form.

MisterGlass, 2023-12-15:
I haven't tested, but I would expect accessibility issues here as well.

MikeDabrowski, 2023-12-29:
I think my chrome finally was able to remember the pwd for this page. Maybe try that.

floretan, 2024-02-06:
Chrome on the desktop remembers basic authentication passwords, but chrome mobile doesn't (tested on a mac and iphone respectively).

lggomez, 2024-02-23:
The issue lies on the frontend expecting basic HTTP auth via the [express-basic-auth](https://github.com/LionC/express-basic-auth) middleware. It would be great to have an UI login but it will require non trivial work to get there

ondratu, 2024-02-23:
We know about problems with some password managers. But without TLS, which is out of chance to implement on Buddy printers, **HTTP Digest**, that is what we actually used, is more secure, that simply sending password in plaint text form, which standard HTML login forms do.

Another way should be creating some type of JavaScript HTTP Digest which will be only because password managers, resp. browsers ignore storing old, but standard authentication mechanisms :-(

We left this issue opened, but please, don't wait any improvements in this case.

MikeDabrowski, 2024-02-23:
What about disabling it all together? Via a special config file on a card for example - will be more hidden than having it in the settings. We are talking about the local network users here, maybe even most of them. I, as one of them, am really fine with leaving my printer open to my local network without password. To open the prusalink you need to type in the ip address, which for most other users of my local wifi is already hacking.

I also feel a bit let down by the fact that in order to flash custom prusalink (in which I would disable this 'feature') I need to permanently break the board (appendix) - its agains my philosophy of open software. I get that it is a kind of coverage for warranty claims but I feel like better options could be worked out here.

MisterGlass, 2024-02-23:
I would also like an option to disable the password. I don't require a password for people to print on my 2d printer and I'd like the same setup for my 3d printer

lggomez, 2024-02-23:
> I would also like an option to disable the password. I don't require a password for people to print on my 2d printer and I'd like the same setup for my 3d printer

I get the rationale (and support having this option in a trusted network) but that's a poor comparison; 2D printers don't have an attack vector that can induce mechanical failures

MikeDabrowski, 2024-02-23:
That is true, however I can imagine that 1- current security also has its own vulnerabilities and 2- if attacker already has access to your internal network you are already in trouble and granted 3dprinter could start a fire, but so can other smart devices hooked up to your lan. 

The bottom line is a matter of having a choice.