prusa3d/Prusa-Link-Web #471: 409:Conflict - File is busy

clivius, 2023-12-25:
Uploading a file to my MK4 (latest firmware and software) also loads the file to be printed on the printer rather than just uploading to the USB drive.  The display on the printer has the file loaded waiting for me to select print.  This is annoying cuz it locks the file from being deleted in Prusalink.  When I'm uploading, I'm never interested to immediately print.  I might want to add more files or reupload the same file cuz I missed a setting or decided to change the slicing output.  It's not a problem to upload more files just the initial upload is in a ready to print state and can't be deleted.  My printer is in another building and a pain in the ass to clear the screen on the printer.

Please, either remove this behaviour so when just uploading, the file is not loaded in a ready to print state.  I realize this is a convenience for printing an uploaded only file but it prevents Prusalink from performing any actions on the file for others.  What if I uploaded a bunch of files and wanted to print anything but the first file I uploaded?

I think if you just uploaded, you're not interested to also have the file loaded on the printer, in a ready to print state.

Thanks/Livius

LoakAudio, 2024-05-08:
Hi,
Using an MK3.5 last firmware (6.0.0), same behavior and I agree with Clivius.
Maybe this behavior can be kept in the dashboard tab, but uploading a file into the projects tab should not trigger the printer to select the file.
Thanks

theta-3d, 2024-12-18:
I found a workaround about this problem:
Send a new file to print, start printing, then stop printing immediately. Nothing printed actually.
After that, you can delete whatever you want without going to printer...

Xargas, 2025-01-30:
I can confirm that this is still an issue on MK4S, Firmware 6.1.3+7898.
Thanks for the workaround hint!