prusa3d/Prusa-Link-Web #473: Prusalink web address based auto login shows Prusalink form, but no data.

VH-fixer, 2024-01-03:
When I use an autologin line on the web browser, Prusalink connects, but shows no data.  No files are shown in storage.  No feedback in the dashboard.  Storage is blank and will not take an upload.    As if the login was not finished.

http://maker:paswwwooord@192.168.01.111 is an example of the login line.
MK4 firmware 5.1.2
Windows 11
Chrome Browser
Asus network router RT-AX82U

I have noticed this at least since Firmware 5.0.  
Picture of the storage option, showing no files.   There are many files on the USB.
[image]

TojikCZ, 2024-01-03:
Huh, never tried this, you seem to be doing funny stuff with the address bar. Anyway, moved the issue to Prusa-Link-Web as I think the (faulty?) code is here

radium-v, 2024-02-03:
I just ran into this issue - I'm trying to use the preview/progress graphic in a browser source in OBS Studio. This browser source doesn't seem to have the ability to display the login popup, so I tried the alternative method of `http://<username>:<password>@<ip address>` with no luck. When I try to access it this way in Chrome, the page does load, but I get this error in the console:

```
Uncaught (in promise) TypeError: Failed to execute 'fetch' on 'Window': Request cannot be constructed from a URL that includes credentials: /api/v1/info
    at main.2812ca58adec020e163e.js:1:75752
```

blester125, 2024-02-14:
I have this same issue, when firefox restarts you need to re-log in. I wanted to be able to have a link (in my homelab dash board) that auto-logs in.

When I use a link with the username and password firefox has a pop-up that says you are providing a log in for a website that doesn't need it. If you say ok to the message you end up at prusa link but with the same behavior mentioned above.

[image]

It looks like there should be some code in the section that parses the username/password from the login box that first checks of the username/password are in the url. If they are it should be extracted and the base url saved for use in later things like the fetch call that is currently throwing the error.

I'd try to make a PR fixing this but I'm not great at js and I'm not sure the best way to test this behavior, is there a way to get the printer to serve a dev js page?

VH-fixer, 2024-08-29:
This is still a problem in 61.2, A fix is more important since we can now assign a hostname to the printer and auto login for a local host is not supported in some browsers.

 Prusa is now entering the market where many people are not "techies".   There is no standard for setting a static IP address in a router. Most people I have worked with have a difficult time or fail at trying to set a static IP in their router.  We cannot rely on  IP "license" timeout.  A router reset, power out or just leaving the printer off for a week or two can change the IP address being used.  Disabling logins and PrusaSlicer connections to the printer when IP addresses are used. 

Using the hostname  frees us from using an IP address.  Simplifing network setup for ALL users.   Example  My MK4 is set to the hostname MK4-hub1.    I can use Http://MK4-hub1 as the URL for a physical printer in Prusa-Slicer.   It works well loading the MK4 from Prusa slicer.  

When I use http://MK4-hub1 in chrome it connects and asks for a login/pw.    I found I CAN NOT set an auto login/pw fill in with Chrome. Chrome requires a "." in the address.  Chrome will default to .com if you don't have a "." in the address.    A "." cannot be used in the hostname.   Autofill will not work for a local host such as http://MK4-hub1

The  choice for auto login using the hostname is http://user:pw@MK4-hub1    This needs to work correctly on PrusaLink.  

We need to use the hostname function and have the auto login line work with PrusaLink.   Not everyone should be forced to use Prusa-connect or an IP address instead.

maximweb, 2024-10-04:
I have the same issue on MK4, just upgraded to Firmware 6.1.3.

Emptying the browser cache seemed to fix it for now. Will keep observing if it starts to be a regular thing...

beegmouse, 2024-11-01:
+1 on this as an issue. I thought the user:pw@IP could fix the autologin with a bookmark, but it doesn't work.

I'm guessing they don't use prusalink at the factory, as manually typing in there pws for multiple printers is a nightmare.

tedder, 2025-03-19:
aha! now I know why I'm seeing this error.