prusa3d/Prusa-Link-Web #476: File explorer bgcode

Fluffybunny96, 2024-01-15:
When you upload a new file to the printer prusalink doesnt recognise the new file format BGCODE.

[image]

To resolve this issue you cannot drag and drop bgcode, you have to browse to a file and then manually change file type to "All files"
[image]

TojikCZ, 2024-01-15:
Note: PrusaLink for MK3 won't support it, but yea, Buddy based printers should allow uploads of those

ondratu, 2024-01-15:
This is duplicity for https://github.com/prusa3d/Prusa-Link-Web/pull/474#issuecomment-1891641230 Buddy printers are fixed yet.

closed 2024-01-15 by ondratu