prusa3d/Prusa-Link-Web #485: [Prusa Connect] (Feature Request) Organize Folders & Files with "drag & drop" / with "Select" --> "Move to"

premiumcola, 2024-03-20:
hi,

there is no function to move & select some files to folders both in (printer files) OR (Connect files) ?!
I would like to store/sort good gcodes for the future in some folders - but there is no function to do so.

In the best case that could simply be done by drag an drop directly in the browser. (printer files)
- Why selecting is not available here?
[image]

Or by adding that into the menues when selecting the files: (Connect files)
[image]

Hint: It would be awesome to do that with prints which are already online.

Thanks a lot & Happy printing!

______________________________
MK4 - 5.1.3+13503 - Buddy SN 10589-3742441631707515

mix579, 2024-03-20:
FIle management really is PrusaConnect's weakest point. Inconsistencies between local and cloud file maagement capabilities. Lack of basic functions like folders in PrusaConnect and drag and drop operations.

BenCos17, 2025-01-03:
this is something I'd want also tbh 
It's one of the main issues that I have with it atm for files management purposes