prusa3d/Prusa-Link-Web #489: Mk3.5 connect shows einsy rambo not buddy board

cozmo14047, 2024-04-22:
Hi, when u connect the mk3.5 to p4usa connect and go to settings it says einsy rambo serial number instead of buddy board serial number

ondratu, 2024-04-23:
Hi, that is not (or can't be) exactly possible. Did you use same printer token as you used in MK3 for Connect communication? If not, what you see is buddy board serial number.

cozmo14047, 2024-04-23:
> Hi, that is not (or can't be) exactly possible. Did you use same printer token as you used in MK3 for Connect communication? If not, what you see is buddy board serial number.

Yes i see the buddy board serial number but it says seinsy rambo sn: (buddy board sn) 
[image]

ondratu, 2024-04-25:
I see, bad label, thanks for report!

cozmo14047, 2024-04-25:
all good, sorry if i was confusing before

> I see, bad label, thanks for report!