prusa3d/Prusa-Link-Web #496: [FEATURE REQUEST] Network cam integration in dashboard

dl9sec, 2024-06-12:
Hi,

it would be great to have the possibility to integrate a network cam stream into the dashboard of the MK4 PrusaLink (similar to the control window of Octoprint).

E.g. if a checkbox "Integrate webstream in dashboard" is set and a link like "http(s)://mycam.local:8083/stream.mjpg" is given, the stream is shown beside the tempertaure chart on the dashboard.

So the MK4 integrated webserver doesn't need resources for streaming, just the client browser...

Regards, Sten

dl9sec, 2024-06-12:
Thank you for the response and moving the issue. I wasn't aware, that the  Prusa Link stuff has its own repo and issue tracker, so I assumed it is part of the buddy-firmware...

alebre-lithoz, 2024-06-25:
+1 for this feature
we have a fleet of prusa printers but cannot use prusa connect because we require a local service and if the prusalink dashboard would allow a camera stream we would have the one feature we are still lacking, and atm using octoprint for.
[image]
this is also already supported in the prusalink version that runs on the raspi Zero that can be added to the Mk3S if I am not mistaking.

m10d, 2024-06-27:
my understanding is that this would be:

- on the `prusa-link-web` side, this seems a low-lift / high ROI for users that can't use Connect.  
    - A new tab (as mocked by @alebre-lithoz ) could contain a react component as simple as https://medium.com/@a.g.stranger/how-to-create-a-react-application-to-stream-your-webcam-basic-beginner-66c792993622 
    - a new config variable to store the connection string to the camera
    - just choose one (or several - but honestly the 3rd party `js` lib you choose will probably handled most common ones out of the box) of the many applicable streaming protocols, say `RTSP`
- users can then simply use `gstreamer`, `v4l2`, or similar readily available software to provide the stream (many commercial cameras support these, too)

the (limited) on-device hardware is **not burdened at all** because the streaming is generated on some other board, and streams directly to the users' browser - all the printer does is serve up a tiny bit of (static) html & js.

m10d, 2024-06-27:
>     * users can then simply use `gstreamer`, `v4l2`, or similar readily available software to provide the stream (many commercial cameras support these, too)

just wanted to add some breadcrumbs for other users to see how they might use this:

- first, if you have a camera accessible on a system (raspberry pi, or any other box) - [this answer](https://raspberrypi.stackexchange.com/a/75763) shows how to stream that camera, and how to test the stream from some client
- here is a deeper primer on v4l https://medium.com/@deepeshdeepakdd2/v4l-a-complete-practical-tutorial-c520f097b590  (note the article doesn't metion it, but it appears[ v4l2 can interact with RTSP](https://www.linux-projects.org/uv4l/tutorials/rtsp-server/) just fine as well)
- on the prusa-web-link side, here's a tutoral on displaying an RTSP feed in reactjs 

note: i have no pet format - I just googled what's the most accepted way these days and RTSP seems to be it - there are other older ways I have used in the past; anything (**standard/open**) that gets the live video feed displayed in the prusa-link web page works.

dl9sec, 2024-06-27:
For me it is the same usecase as for @alebre-lithoz. I have a Octopi running in parallel, currently just to be able to remote monitor the print. The Octoprint camera system offers already a simple MJPG stream (http://octopi.local/webcam/?action=stream).
Maybe also the stream format of nertwork cameras should be supported (I thing RTSP is one of the most common)...

alebre-lithoz, 2024-06-27:
just to clarify my previous comment. this is already available in prusa-link v1 for the Mk3S upgrade with raspberry pi: https://help.prusa3d.com/guide/camera-setup-for-prusalink-prusa-connect_470943
so the functionality should already be there and could maybe ported over to PrusaLink for the Buddy Board and maybe even the Mini if it does only require minimal additional computing power and storage space